tobozo/ESP32-targz #1: Unable to untar more than one file

chaosct, 2020-02-07:
Hi there.

I'm trying to use the `tarExpander` function and it only correctly extracts the first file. Afterwards the function exitswithout error.

I've been trying to understand why this is the case, but it is beyond me. Any ideas?

tobozo, 2020-02-07:
hey thanks for your feedback :-)

can you please specify how many files in your tar file, and what's their size ?

there's an accidental limitation I'm still trying to fix when the file is smaller than the buffer size, doest it still occur with bigger files ?

tobozo, 2020-02-08:
@chaosct thanks to your reporting I just published a fix that should address that issue, can you update your local version of the ESP32-targz library from the master branch of this repository and confirm this is now working for you ?

If so I'll produce a new release and propagate that through the Arduino Library Manager

tobozo, 2020-02-18:
This project has [landed in the Arduino Library Manager](https://github.com/arduino/Arduino/issues/9739).

Closing this issue as solved, feel free to reopen if you feel I've missed something :-)

closed 2020-02-18 by tobozo