tobozo/ESP32-targz #3: Corrupted file path when untarring file

MormonMoron, 2020-06-16:
The file path is getting corrupted in the unTarHeaderCallback because a new char* buffer is allocated, but not null terminated before use.

You should change the snippet (starting at line 349 in the current master branch):

```
int unTarHeaderCallBack(header_translated_t *proper,  CC_UNUSED int entry_index,  CC_UNUSED void *context_data) {
  dump_header(proper);
  if(proper->type == T_NORMAL) {
    char *file_path = new char[256];// = ""; // TODO: normalize this for fs::FS, limit is 32, not 256
    // avoid double slashing root path
    if( strcmp( tarDestFolder, FOLDER_SEPARATOR ) != 0 ) {
      strcat(file_path, tarDestFolder);
    }
    strcat(file_path, FOLDER_SEPARATOR);
    strcat(file_path, proper->filename);
```
to null terminate the file_path buffer after allocation:

```
int unTarHeaderCallBack(header_translated_t *proper,  CC_UNUSED int entry_index,  CC_UNUSED void *context_data) {
  dump_header(proper);
  if(proper->type == T_NORMAL) {
    char *file_path = new char[256];// = ""; // TODO: normalize this for fs::FS, limit is 32, not 256
    file_path[0] = '\0';
    // avoid double slashing root path
    if( strcmp( tarDestFolder, FOLDER_SEPARATOR ) != 0 ) {
      strcat(file_path, tarDestFolder);
    }
    strcat(file_path, FOLDER_SEPARATOR);
    strcat(file_path, proper->filename);
```

tobozo, 2020-06-17:
thanks for spotting that bug !

changes have been published in the [latest release](https://github.com/tobozo/ESP32-targz/releases/tag/0.1.1) and should propagage soon through the arduino library manager

closed 2020-06-17 by tobozo

jpswensen, 2020-06-18:
Thanks! I saw it in the Arduino Library Manager this morning.