tobozo/ESP32-targz #5: error occurs when add library

m2sarmadi, 2020-10-03:
hello everyone
This error occurs as soon as the library is added :

```
C:\Users\Pc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/sdk/lwip2/include/lwip-err-t.h:8:28: error: 's32_t' has a previous declaration as 'typedef long int s32_t'
 typedef signed     long    s32_t;
                            ^
```

tobozo, 2020-10-03:
hello, thanks for your feedback

I tried to reproduce using esp8266 2.7.4 + LittleFS 0.1.0, but it works for me.

What's your package version ?

```log
Executable segment sizes:
IRAM   : 27224   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
BSS    : 31360 )         - zeroed variables      (global, static) in RAM/HEAP 
DATA   : 1272  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 3536  ) / 81920 - constants             (global, static) in RAM/HEAP 
IROM   : 274192          - code in flash         (default or ICACHE_FLASH_ATTR) 
Using library LittleFS at version 0.1.0 in folder: ~/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/LittleFS 
Using library ESP32-targz at version 0.1.2 in folder: ~/Arduino/libraries/ESP32-targz 
~/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-size -A /tmp/arduino_build_814719/Unpack_targz_file.ino.elf
Sketch uses 306224 bytes (31%) of program storage space. Maximum is 958448 bytes.
Global variables use 36168 bytes (44%) of dynamic memory, leaving 45752 bytes for local variables. Maximum is 81920 bytes.
```

m2sarmadi, 2020-10-03:
in ESP32-targz.h , i deleted this lines :
`#include "spiffs/spiffs.h"` and `#include <LittleFS.h>`

then i added this lines :
```
#ifdef USE_LittleFS
  #include <FS.h>
  #define SPIFFS LittleFS
  #include <LittleFS.h>
  #endif
```

and it worked perfectly!

tobozo, 2020-10-03:
thanks for spotting that!

I've pushed the changes and prepare the next release

m2sarmadi, 2020-10-03:
Thanks for your helpful library

closed 2020-10-03 by m2sarmadi