tobozo/ESP32-targz #6: Unable to untar

m2sarmadi, 2020-10-03:
I have a .tar file that contains files with the following size(bit): 403,000 and 728 and 342,548 and 132,795 and etc. (most file formats are .js)
It works fine when I start untaring it, but when it gets to a specific file it gives the following error :
`Could not open [fileName] for write`
And then the untaring of the rest of the files is also ignored.
I am sure that this particular file is not a problem because it is untar alone in another sketch.

tobozo, 2020-10-03:
I can only think of a filesystem limitation here e.g. there's a 32 characters (path+filename) limit to FS.h, or the size of the tar + uncompressed content exceeds the partition size.

How many files are there in this tar archive and how deep is the folder structure in the tar ?

m2sarmadi, 2020-10-03:
22 file are in tar archive
and
this files are in root of tar archive
The total size of the tar aechive is 1.2 M.B
Probably a problem with SPIFFS size

tobozo, 2020-10-03:
yup I just discovered there are many filesystem sizes in the SDK menu :sweat_smile:  

[image]

tobozo, 2020-10-09:
@m2sarmadi did you get better results with smaller archives or should I keep this issue open as a bug ?

Since the WebServer component can serve gz contents, maybe minify+concat+gzip'ing your css/js/html files before creating the tar will work better ? (also using a css sprite image can decrease the amount of files)

tobozo, 2020-11-09:
closing this as more relevant messages are now displayed when the filename exceeds 32 chars.

feel free to reopen if the symptoms are still there for you

closed 2020-11-09 by tobozo