tobozo/ESP32-targz #11: Enhancement request:  Suggestion for failure behavior

scubachristopher, 2020-11-14:
Heya @tobozo,

With at least gzUpdater, perhaps you would consider returning an ERROR code instead of void and quieting the error messages, leaving that to the application?

Chasing down something on my end, and when gzUpdater unexpectedly returns (instead of restart) because of error, I have no information in my application I can send to the operator (Cloud):
```
Attempting to flash
[E][ESP32-targz.cpp:246] gzProcessBlock(): Error uncompressing data
[E][Updater.cpp:264] end(): premature end: res:0, pos:385024/1159168
```

If that makes sense, awesome.  If you want to keep as is, totally fine, I'll just fork.

Happy to do a PR if you agree, but busy moving house starting this AM for a few days.  In fact, I'll do the PR if you move me? ;)

Thanks for the consideration

tobozo, 2020-11-14:
hey @scubachristopher thanks for your feedback

This totally makes sense, do you have an example of faulty gz file I can test with, or point me to a way I can produce such file by myself ?

tobozo, 2020-11-14:
Thanks again for your inspiring feedback, I just pushed those changes, it's now acting as the Update API, returns false in case of error, and provides `tarGzGetError()` for debug.

 - [Better Error Handling](https://github.com/tobozo/ESP32-targz/commit/149a3895a0b1bdc1f5e5fdf072660ea3880f3435)
 - [Updated Examples](https://github.com/tobozo/ESP32-targz/commit/52b8e477a3b33227772108db9fce52d1b6a249a1)

```C
  if( ! gzUpdater(SPIFFS, firmwareFile ) ) {
    Serial.printf("gzUpdater failed with return code #%d", tarGzGetError() );
  }
```

Happy unzipping in your new house ^^

scubachristopher, 2020-11-15:
Looks great!  Appreciate it!  Sent a few Euros so you can get a coffee (or better yet, a beer) on me ;)

Can you bump the version?  For some reason, VS Code still pulls down the old one for me, even after blowing out the .pio directory.

Thanks again man!

closed 2020-11-15 by scubachristopher

tobozo, 2020-11-15:
ooh thanks!

bumped the version to 0.2.1 in the Library Manager  :-)