tobozo/ESP32-targz #16: I cant compile the example code for esp8266

denruss, 2021-01-11:
In file included from C:\Users\Denis\Documents\Arduino\libraries\ESP32-targz\src/ESP32-targz-lib.h:44:0,

                 from C:\Users\Denis\Documents\Arduino\libraries\ESP32-targz\src/ESP32-targz.h:124,

                 from C:\Users\Denis\Documents\Arduino\libraries\ESP32-targz\examples\Unpack_gz_file\Unpack_gz_file.ino:15:

C:\Users\Denis\Documents\Arduino\libraries\ESP32-targz\src/helpers/md5_sum.h:4:25: fatal error: mbedtls/md5.h: No such file or directory

tobozo, 2021-01-11:
thanks for reporting this :+1: 

I just pushed a fix on the master branch, untested though as I can't test on a real device before a couple of days.

denruss, 2021-01-12:
thanks. I was able to compile now

infrafast, 2021-01-13:
Hello Toboyo; wyour lib looks very good, thanks for sharing
 When I compile I get the following error :

C:\Users\progman\arduino\libraries\ESP32-targz\src/helpers/md5_sum.h:4:25: fatal error: mbedtls/md5.h: No such file or directory

I can't find this md5 files dependency:... would you have any advise to help?

Thanks

tobozo, 2021-01-14:
closing this as solved after testing most examples on a Wemos Mini D1 and producing a new release, thanks again for your input :+1: 

feel free to reopen if you get a compilation error with one of the examples

closed 2021-01-14 by tobozo