tobozo/ESP32-targz #18: E][Updater.cpp:155] begin(): too large 1183629312 > 1310720 -- halts ESP32

scubachristopher, 2021-01-12:
Hi,

What's interesting is I've had no issues with this code for some time.  Download was successful, file size was 673935.  Once I download the new firmware to SPIFFS, I reboot the ESP and immediate try to flash.  When I call:

gzUpdater(SPIFFS, "/dlfirmware.bin.gz");

I get:

[E][Updater.cpp:155] begin(): too large 1183629312 > 1310720

Fine.  Something went wrong.  But it literally halts the ESP.  Am moving my wdt start above this code -- perhaps that will solve the halt but reproducing this might be difficult.

I am not familiar with how the update process works (begin, end), but is there a size check that can gracefully catch this scenario in gzUpdater?

Any perspective on this would be appreciated.  Halting on FW update will brick the device, as I cannot remove the fw from SPIFFS until after gzUpdater returns, which it did not...

tobozo, 2021-01-12:
hey Christopher,

It sounds like `tarGzStream.output_size` has an invalid value and has the Update process fail.

Obviously tarGzGetError() is useless after a system halt, so re-enabling the logging could help identify a possible error during decompression, could be a consequence of a nearly full or damaged SPIFFS partition, or a bug depending on the version of the library you're using.

I'll need to know what version of the library produced this error though, there's a possibility it will behave differently with the latests release (0.3.0) or with the master branch (0.3.1).

The [gzUpdater example sketch](https://github.com/tobozo/ESP32-targz/blob/master/examples/Update_from_gz/Update_from_gz.ino) was recently updated, it was tested both on M5Stack and LoLinD32 pro successfully.

Can you try and see if this one works with the master branch ? This is the expected output:

[image]

scubachristopher, 2021-01-12:
I think I understand the issue, but do not know how to solve.  I decided to show what files exist and I found part of my firmware still on the file system.  Fine, put in a delete in case there was an error to cleanup.  But now:

```
===== File system info =====
Total space:      1378241 bytes
Total space used: 494219 bytes

Total space free: 884022 bytes

FILE: /httpcert.pem.der Size: 1011
FILE: /index.lrigz Size: 52446
FILE: /client.crt.der Size: 894
FILE: /httpcert.pem Size: 1424
FILE: /client.key.der Size: 1191
FILE: /ca.crt.der Size: 1011
FILE: /ca.crt Size: 1424
FILE: /client.crt Size: 1265
FILE: /client.key Size: 1675
```

My code:
```

  unsigned int totalBytes = SPIFFS.totalBytes();
  unsigned int usedBytes = SPIFFS.usedBytes();

  Serial.println("\n\n===== File system info =====");
  Serial.printf("Total space:      %d bytes\n", totalBytes);
  Serial.printf("Total space used: %d bytes\n\n", usedBytes);
  Serial.printf("Total space free: %d bytes\n\n", totalBytes - usedBytes);

  File root = SPIFFS.open("/");
  file = root.openNextFile();
 
  while (file) {
      Serial.printf("FILE: %s Size: %d\n", file.name(), file.size());
      close(file);
      file = root.openNextFile();
  }
  close(root);

```

So, that file list just doesn't add up to 494,219 bytes.  I don't see any other files.  My concern now is there are some temporary files that are hidden or the decompression somehow mangled the SPIFFS file system.

I am hesitant to touch the SPIFFS partition now, because I need to understand where the space has gone.  So I can't run the example at this time.

This is a standard 4MB ESP32 Dev kit, with a standard SPIFFS partition definition.

Can you provide any insight as to temp files created during the decompression?

tobozo, 2021-01-12:
With gzUpdater there is no temporary file, the decompressed bytes are passed directly to Update::writeStream and written on the OTA partition.

SPIFFS has a few annoying fragmentation bugs though, most of them are triggered by writing incomplete files, overwriting or deleting, all solved by formatting.

`tarGzExpander` can produce exactly those symptoms when using an intermediate file that nearly fills the partition.

Here's how to use tarGzExpander without the intermediate file:

    tarGzExpander(SPIFFS, "/certsuite.tar.gz", SPIFFS, "/", nullptr );

This won't prevent files from being overwritten but may avoid being close to a nearly full partition, so it's more a mitigation than a solution.

LittleFS has a better error detection and is much more resilitent, I wish this was the official partition scheme instead of SPIFFS :-)

scubachristopher, 2021-01-13:
First, thanks for the attention to this issue.  I do download the firmware to a fixed file name, process then delete the file.  Rinse and repeat.

If SPIFFS has some fundamental issue writing to a filename that is constant, then deleting and potentially ending up with internal fragmentation, that's a show stopper for me because I preload the certs / keys for secure connections in the SPIFFS partition.  The downloaded firmware sits along side that content (about 60k).  If at some point the available space < gzipped firmware, I cannot simply reformat.

I'll have to look into this deeper, perhaps multiple SPIFFS partitions, one for static data and one for downloads that can be reformatted.

Since this intersects with your work, I'll inform where I end up with this so others can understand at least where I land on this.

Many thanks again

tobozo, 2021-01-13:
if that may be of interest, I'm currently researching how to do http-stream => gzuncompress => update-partition with no intermediary file at all, but the simple example sketch is already eating 67% of program storage space so it's not really lightweight, and I still have to make it work ^^

scubachristopher, 2021-01-14:
Dude.  That would be killer!

tobozo, 2021-01-14:
Ive added a [gzStreamUpdater](https://github.com/tobozo/ESP32-targz/blob/master/examples/Update_from_gz_stream/Update_from_gz_stream.ino) example to the latest release but it makes a huge sketch, I wonder how this will affect projects with existing WiFiClient and http objects, maybe not that much.

scubachristopher, 2021-01-14:
Wow -- awesome.  Let me give it a shot!

scubachristopher, 2021-01-15:
So here are the results of the test of this new capability with my rig.  As per your sketch, once my http connection is established, I just injected the gzStreamUpdater call ahead of what I normally do:

```
        ... // setup done above this code

        file = SPIFFS.open("/dlfirmware.bin.gz", FILE_WRITE);
        httpCode = http.GET();

        if (httpCode > 0) {

// NEW
          if (!gzStreamUpdater(http.getStreamPtr(), UPDATE_SIZE_UNKNOWN) ) {
            Serial.printf("gzHTTPUpdater failed with return code #%d", tarGzGetError());
          }
// END OF NEW

          // get length of document (is -1 when Server sends no Content-Length header)
          len = fw_size = http.getSize();
          Serial.printf("HTTP.get successful for %s, size of file reported as %d\n", buff, fw_size);
          // create buffer for read
          uint8_t buff[1024 * 16] = { 0 };

          // get tcp stream
          WiFiClient *stream = http.getStreamPtr();

          // read all data from server
          while (http.connected() && (len > 0 || len == -1)) {
              esp_task_wdt_reset();
              size_t streamSize = stream->available();
              if (streamSize) {
                  int c = stream->readBytes(buff, ((streamSize > sizeof(buff)) ? sizeof(buff) : streamSize));
                  Serial.printf("Read %5d bytes from stream, %6d remaining\r", c, len);
                  file.write(buff, c);
                  if (len > 0) {
                      len -= c;
                  }
              }
          }
          file.close();
        }
        else {
          Serial.printf("HTTP.get UNSUCCESSFUL for %s\n", buff);
        }
        ...
```

Running with -DCORE_DEBUG_LEVEL=5, there's a per-character log line emitted from line 431.  I strongly suggest you remove this, else it's a per-character read log line and life is too short:
```
log_v("read 1 byte: 0x%02x", out[0] );
```
In order to really see what's going on, I commented that out.  Much better log output:

```
[V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 196380
[V][ssl_client.cpp:58] start_ssl_client(): Starting socket
[V][ssl_client.cpp:93] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:102] start_ssl_client(): Setting up the SSL/TLS structure...
[V][ssl_client.cpp:115] start_ssl_client(): Loading CA cert
[V][ssl_client.cpp:180] start_ssl_client(): Setting hostname for TLS session...
[V][ssl_client.cpp:195] start_ssl_client(): Performing the SSL/TLS handshake...
[V][ssl_client.cpp:216] start_ssl_client(): Verifying peer X.509 certificate...
[V][ssl_client.cpp:225] start_ssl_client(): Certificate verified.
[V][ssl_client.cpp:240] start_ssl_client(): Free internal heap after TLS 157736
[D][HTTPClient.cpp:1025] connect():  connected to test.lagunaresearch.com:443
[V][ssl_client.cpp:279] send_ssl_data(): Writing HTTP request...
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'HTTP/1.1 200 OK'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'server: ecstatic-3.3.2'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'cache-control: max-age=3600'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'last-modified: Thu, 14 Jan 2021 21:41:43 GMT'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'etag: W/"256508-673933-2021-01-14T21:41:43.126Z"'
[... 614 lines cut ...]
HTTP.get successful for https://test.lagunaresearch.com/chris_cloud.bin.gz, size of file reported as 673933
Read     4 bytes from stream, 673933 remaining
```

So a few observations:
1).  The size of the decompressed file is:
```
pi@test:~/lri_firmware$ gzip -l chris_cloud.bin.gz
         compressed        uncompressed  ratio uncompressed_name
             673933             1199680  43.8% chris_cloud.bin
```
which means it's not reading the entire file.  Once the fail happens, it falls through to my original loop which is able to read an additional 4 bytes from the stream, and that would be the complete read.

2).  I see these occasionally during the download:
```
[E][ESP32-targz-lib.cpp:428] readSourceByte(): readSourceByte read error
```

3). The number of read errors does NOT correspond to lost bytes.

4). Multiple runs always end with the same 4 bytes not read from the stream.

If this works, it's a FAR superior method of pulling firmware, so I'm excited to get through this.

Finally, my firmware is just that -- there is no 4-byte header for size or anything abnormal about staging this on the server.  Just a gzip --keep --best --force <firmware> on an ubuntu server.

tobozo, 2021-01-15:
thanks for this detailed report

I'm not sure where those 4 missing bytes come from, and I can't reproduce this error so I'll assume it's a consequence of the previous `readSourceByte` errors.

Because the `http.getSize()` may return -1 depending on the remote webserver (Content-length is optional), the gzip size has been made irrelevant to the stream decompression process.

The message `gzUncompress(): decompressed 1199676 bytes` is reporting a wrong value anyway, it doesn't take the zerofills into account (fixing that now).

Since I'm in the middle of researching `tarGzStreamExpander( Stream *stream, fs::FS &destFS, const char* destFolder )` that fix has landed on the [tarGzStreamExpander](https://github.com/tobozo/ESP32-targz/tree/tarGzStreamExpander) branch.

scubachristopher, 2021-01-16:
Some good news!  This consistently works:
```
// consume and return a byte from the source stream into the argument 'out'.
// returns 0 on success, or -1 on error.
static unsigned int readSourceByte(struct TINF_DATA *data, unsigned char *out)
{
  //if( !tarGzStream.gz->available() ) return -1;
  if (tarGzStream.gz->readBytes( out, 1 ) != 1) {
    log_e("readSourceByte read error, available is %d.  attempting one-time retry", tarGzStream.gz->available());
    if (tarGzStream.gz->readBytes( out, 1 ) != 1) {
    log_e("readSourceByte read error, available is %d.  failed at retry", tarGzStream.gz->available());
      return -1;
    }
  } else {
    //log_v("read 1 byte: 0x%02x", out[0] );
  }
  return 0;
}
```
So, aside from killing the log_v read 1 byte (I beg you), just add a one-time retry.  Also, if this routine does ever error out, I'd abort the attempt.

Now when it fails, I only see the initial fail fingerprint in the log:
```

[V][ESP32-targz-lib.cpp:537] gzUncompress(): [INFO] Buffer full, now writing 4096 bytes (total=188416)
[V][ESP32-targz-lib.cpp:202] gzUpdateWriteCallback(): Wrote 4096 bytes
[E][ESP32-targz-lib.cpp:430] readSourceByte(): readSourceByte read error, available is 0.  attempting one-time retry
[V][ESP32-targz-lib.cpp:537] gzUncompress(): [INFO] Buffer full, now writing 4096 bytes (total=192512)
...
[V][ESP32-targz-lib.cpp:202] gzUpdateWriteCallback(): Wrote 4096 bytes
[V][ESP32-targz-lib.cpp:537] gzUncompress(): [INFO] Buffer full, now writing 4096 bytes (total=1167360)
[V][ESP32-targz-lib.cpp:202] gzUpdateWriteCallback(): Wrote 4096 bytes
[E][ESP32-targz-lib.cpp:430] readSourceByte(): readSourceByte read error, available is 0.  attempting one-time retry
[V][ESP32-targz-lib.cpp:537] gzUncompress(): [INFO] Buffer full, now writing 4096 bytes (total=1171456)
[V][ESP32-targz-lib.cpp:202] gzUpdateWriteCallback(): Wrote 4096 bytes
[V][ESP32-targz-lib.cpp:537] gzUncompress(): [INFO] Buffer full, now writing 4096 bytes (total=1175552)
[V][ESP32-targz-lib.cpp:202] gzUpdateWriteCallback(): Wrote 4096 bytes
[V][ESP32-targz-lib.cpp:537] gzUncompress(): [INFO] Buffer full, now writing 4096 bytes (total=1179648)
[V][ESP32-targz-lib.cpp:202] gzUpdateWriteCallback(): Wrote 4096 bytes
[V][ESP32-targz-lib.cpp:537] gzUncompress(): [INFO] Buffer full, now writing 4096 bytes (total=1183744)
[V][ESP32-targz-lib.cpp:202] gzUpdateWriteCallback(): Wrote 4096 bytes
[V][ESP32-targz-lib.cpp:537] gzUncompress(): [INFO] Buffer full, now writing 4096 bytes (total=1187840)
[V][ESP32-targz-lib.cpp:202] gzUpdateWriteCallback(): Wrote 4096 bytes
[V][ESP32-targz-lib.cpp:537] gzUncompress(): [INFO] Buffer full, now writing 4096 bytes (total=1191936)
[V][ESP32-targz-lib.cpp:202] gzUpdateWriteCallback(): Wrote 4096 bytes
[V][ESP32-targz-lib.cpp:202] gzUpdateWriteCallback(): Wrote 3648 bytes
[V][ESP32-targz-lib.cpp:202] gzUpdateWriteCallback(): Wrote 448 bytes
[D][ESP32-targz-lib.cpp:580] gzUncompress(): decompressed 1199680 bytes
[D][ESP32-targz-lib.cpp:687] gzStreamUpdater(): OTA done!
[D][ESP32-targz-lib.cpp:690] gzStreamUpdater(): Update finished !
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 15 - AP_STACONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 15 - AP_STACONNECTED
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
```

It's unclear why there is a read fail and the available() is zero, but next attempt it always succeeds.  I've put a delay(1) in that routine before and tested, but it was exactly the same behavior.  What might make sense is to give it 5 attempts with a delay(1) after the fail, then flag the entire process should bail if it can't read.  Again, it always succeeds the second attempt for me.

So yay!  This should work, and I believe it's a superior approach to firmware updates on the esp32 (gzipped fw, download to a 4k buffer, decomp to the ota partition).  Awesome!

tobozo, 2021-01-16:
ooh thanks for finding that out, I didn't know it was even possible to recover from a single read error ;)

I've added the changes you suggested to my current workspace and will commit them soon with many other changes : I'm namespacing TAR and GZ underlying libraries and moving from functions to object+methods.

Those are breaking changes and I'll probably have to increment the library major version number and add the alpha flag, update documentation+examples, and also add an extra development iteration for ESP8266 support before I publish the next release, so it'll stay on a [separate branch](https://github.com/tobozo/ESP32-targz/tree/refactor) until then.

tobozo, 2021-02-15:
hey @scubachristopher I want to close this thread as it's getting very long, can you confirm the original issue is now fixed for you?

The new release of esp32-targz includes your suggested fix.
Additionnally you may be interested by the new tarGzStreamUpdater function inspired by the esp32OTA thread.

scubachristopher, 2021-02-16:
heya @tobozo:
1) Confirmed GZUnpacker->gzStreamUpdater(http.getStreamPtr(), UPDATE_SIZE_UNKNOWN) works perfectly.
2) What advantage would tarGzStreamUpdater provide over gzStreamUpdater?

Thanks (and will close)

tobozo, 2021-02-16:
thanks!

Actually tarGzStreamUpdater is only useful when both spiffs and app partitions need to be flashed.

It's slightly faster since it saves one extra HTTP connection, maybe more reliable as some situations won't let the ESP make two HTTPS connections in a row.
Also contents of the .tar.gz file can eventually have interesting files such as esp32FOTA's firmware.json, a throwable encryption key, a script to update NVS, a second binary for the other OTA partition, and so on.

closing this issue as solved

closed 2021-02-16 by tobozo