tobozo/ESP32-targz #28: 'tarGzFs' was not declared in this scope

frankcohen, 2021-03-05:
I am using Arduino IDE 1.8.13 on MacOS 11.2.1. I have this in a sketch:

// ESP32-targz
#define DEST_FS_USES_SD
#include <ESP32-targz.h>

boolean tarDecompress()
{
      tarGzFs.begin();
      . . .
}

/Users/frankcohen/Documents/Arduino/ShowRunner/ShowRunner.ino: In function 'boolean tarDecompress()':
ShowRunner:512:7: error: 'tarGzFs' was not declared in this scope
       tarGzFs.begin();
       ^

Bet I'm doing something dumb. Ideas?

frankcohen, 2021-03-05:
tarGzFs is defined as SD in the .h file. I already do SD.begin() in the setup() method. I commented out tarGzFs.begin() and it complies.

tobozo, 2021-03-05:
Hey Frank thanks for reporting this.

Looks like there's some inconsistency in the documentation, the filesystem object should be inherited as `tarGzFS` instead of `tarGzFs`.

I'll update the ReadMe accordingly, thanks for spotting that !

frankcohen, 2021-03-05:
Glad to help. Thank you for your effort.

tobozo, 2021-03-18:
documentation is fixed in the latest release (currently propagating), closing this issue as solved, thanks again for your feedback :+1

closed 2021-03-18 by tobozo