tobozo/ESP32-targz #38: License issues

javipelopi, 2021-04-28:
Hi!

This is a great work and I hope you continue to develop it even further!

However, there might be issues if you want to use this commercially, as it is more than probable that you would need to use the GPL license for your work as you can see discussed [on stackexchange](https://opensource.stackexchange.com/questions/7078/is-it-legal-to-use-gpl-code-in-a-proprietary-closed-source-program-by-putting-i)

I would recommend doing two separate libraries, one for GZ with the MIT license and another with everything TAR GZ, TAR, GZ with the GPL license.

Anyway, I might be wrong, I am just letting you know as the legal part is usually ignored...

Keep the good job!

tobozo, 2021-04-28:
hey @javipelopi thanks for your feedback

According to [stack exchange](https://softwareengineering.stackexchange.com/questions/304874/declaring-multiple-licences-in-a-github-project) and other sources it's not a problem to do that as long as each imported folder has its own licence file and the main licence file is referencing them.

So the GPL can be resumed as: "You are not allowed to modify the code and sell it as a commercial product without also releasing the source code.", which is the purpose of this git repository.

If you need to modify this library for commercial use, the only legal obligation is to release the modified code for the GPL part (in the tinyUntar folder).

TinyUntar: GPL (see https://github.com/tobozo/ESP32-targz/blob/master/src/TinyUntar/LICENSE)
uzlib: zlib (see https://github.com/tobozo/ESP32-targz/blob/master/src/uzlib/LICENSE)
Rest of the code: MIT

Please let me know if you still think this is a legal issue, if so I'll convert this thread to a discussion topic,

javipelopi, 2021-04-28:
Hello @tobozo ,

That is indeed correct. The source code is available so you have done your part.

I am not a lawyer so I have no definitive answers as copyright is such a muddy topic, I will close this issue as it is not code related and you have already solved the question.

Great job!

closed 2021-04-28 by javipelopi