tobozo/ESP32-targz #43: decompress to PSRAM instead of spi flash option

rjjrbatarao, 2021-07-20:
I found your library useful for webserver application, I wanted to implement template processing on gzipped html files, the decompression works in spi flash/ sd but i wanted to decompress it on psram temporarily then feed it to template processing of the web server, this way could prolong spi flash, make html file with template smaller and utilize the external ram of esp32 like esp32 wrover or ttgo boards

tobozo, 2021-07-20:
hi, thanks for your feedback :+1: 

ESP32-targz supports PsRamFS already, you may enable it as follows:

```C

#define DEST_FS_USES_PSRAMFS 
#include <PSRamFS.h>
#include <ESP32-targz.h>

```

Also PSRamFS is VFS compatible so it is mountable on a WebServer as well as usable with ESP32-targz

```C
  server.serveStatic("/", PSRamFS, "/");
```

Please let me know if some features are missing,

rjjrbatarao, 2021-08-04:
Awesome thanks for this great library.

closed 2021-08-04 by rjjrbatarao