tobozo/ESP32-targz #48: Empty file inside .tar archive returns error -34

JamPaul97, 2021-09-30:
If there is an empty file inside a .tar file when trying to extract it the process exit with code -34. A log file can be provided if needed.

tobozo, 2021-09-30:
hey @JamPaul97 thanks for reporting that.

I'll try to reproduce the issue and post my folluwup here.

tobozo, 2021-09-30:
update: thanks to your feedback I could reproduce and [fix the issue](https://github.com/tobozo/ESP32-targz/commit/d2d25b5b9f512773dda5ce0ad2f4b32f04c7ce20).

I'll produce a new release soon.

tobozo, 2021-09-30:
[New release available](https://github.com/tobozo/ESP32-targz/releases/tag/v1.1.2), it will propagate soon to arduino and platformio library registries.

Closing this as solved, thanks again for your feedback :+1:

closed 2021-09-30 by tobozo