tobozo/ESP32-targz #52: uncompress tar stream (file or HTTP) to any destination

mc-hamster, 2022-01-02:
Hi!

I see methods to uncompress .tar.gz and .gz streams but no existing methods to uncompress just tar stream and save it to the file system. Is there a way to do this with the library now?

Thanks!

tobozo, 2022-01-02:
hey, thanks for your feedback.

Indeed there was no stream method for tarExpander, I've fixed that on branch 1.1.4 and will produce a new release soon.

Thanks for pointing out the lack of stream support for tar :+1:

tobozo, 2022-01-02:
there's a [new release](https://github.com/tobozo/ESP32-targz/releases/tag/v1.1.4) implementing tarStreamExpander, it'll propagate soon in Arduino registry and is already available in platformio registry.

```C++
    bool tarStreamExpander( Stream *stream, size_t streamSize, fs::FS &destFS, const char* destFolder );
```

I'm closing this issue, feel free to reopen and/or add followup.

closed 2022-01-02 by tobozo

mc-hamster, 2022-01-02:
just wow, thanks mate!

mc-hamster, 2022-01-02:
@tobozo 

I updated an example with the new method and can't seem to have it write to the file system.

https://github.com/mc-hamster/ESP32-targz/tree/master/examples/ESP32/Unpack_tar_gz_stream

Could I kindly ask that you look this over? It's probably something glaringly obvious.

I'll send a PR for this example after some cleanup.

tobozo, 2022-01-02:
All I can think of is it doesn't like `-1` as a stream size.

If the web server you're downloading the file from sends the `Content-Length` headers you can get the size as follows:

```C++
      String contentLengthStr = http.header("Content-Length");
      contentLengthStr.trim();
      int64_t streamSize = -1;
      if( contentLengthStr != "" ) {
        streamSize = atoi( contentLengthStr.c_str() );
      }
      if (!TARUnpacker->tarStreamExpander(streamptr, -1, tarGzFS, "/"))
      // ...

```

mc-hamster, 2022-01-03:
@tobozo Rock on! I'll submit a PR for the example to support this shortly.