tobozo/ESP32-targz #56: Readme document outdated

ryanhz, 2022-05-11:
In the readme document:

`if( ! GZUnpacker->gzUpdater(tarGzFS, firmwareFile, /*don't restart after update*/false ) ) {
     Serial.printf("gzUpdater failed with return code #%d\n", GZUnpacker->tarGzGetError() );
}`

Should be changed to

`if( ! GZUnpacker->gzUpdater(tarGzFS, firmwareFile, U_FLASH, /*don't restart after update*/false ) ) {
     Serial.printf("gzUpdater failed with return code #%d\n", GZUnpacker->tarGzGetError() );
}`

tobozo, 2022-05-11:
hey thanks for pointing that out :+1: 

I'm not sure ESP8266 and ESP32 use the same macros for flash/spiffs partitions though, will test that later and edit accordingly.

ryanhz, 2022-05-11:
Thanks for your prompt reply.
Actually it is not a micro, it is a function in GzUnpacker class:
`bool    gzUpdater( fs::FS &sourceFS, const char* gz_filename, int partition = U_FLASH, bool restart_on_update = true ); `

ryanhz, 2022-07-01:
Thank you for your update and this amazing library. Closing this issue now.

closed 2022-07-01 by ryanhz