tobozo/ESP32-targz #60: Can this be called from web async callback?

jamesarm97, 2022-10-13:
I was trying to uncompress a file after being uploaded using asyncweb server on ESP32 inside the request callback when the file has finished uploading, but during the uncompress (I am using default progress so I see it get to about 40%), I am seeing a watchdog and the board resets. Is there a proper way or am I missing something? This is what is being logged:

E (81627) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (81627) task_wdt:  - async_tcp (CPU 0/1)
E (81627) task_wdt: Tasks currently running:
E (81627) task_wdt: CPU 0: async_tcp
E (81627) task_wdt: CPU 1: IDLE
E (81627) task_wdt: Aborting.

tobozo, 2022-10-13:
Hi, thanks for your feedback :+1: 

The watchdog being called sounds like a common AsyncTCP pitfall where the sketch is doing blocking stuff from an asynchron context (the request callback).

The ususal solution is to register the request data (e.g. path to the file) when upload is complete, and defer the synchron work (decompression) to the main loop.

Another solution is to run the decompression from the "other" core using a FreeRTOS task with idle priority that you spawn from inside the request callback.

jamesarm97, 2022-10-13:
If a .gz file contains many files and want to extract to root of a littlefs partition, can I use "/" as the destination? Its description is destination file and when I tried it, it said something about removing / failed and got a hard lockup after that. I noticed littlefs header uses /littlefs as a "base path", what ever that is.

tobozo, 2022-10-13:
> If a .gz file contains many files

a .gz stream can only contain one file so I guess you're asking about 'tar.gz' format, right?

> can I use "/" as the destination?

only with `tarGzExpanderNoTempFile()` otherwise there is a risk of overwriting the tar file.

>  it said something about removing / failed
> got a hard lockup after that.

you can use `yourGzObject->haltOnError( false )` to prevent lockups

also it's safe to ignore errors about directories when using littlefs/spiffs. where directories don't really exist like on a real filesystem.

tobozo, 2023-05-01:
closing this a answered, feel free to reopen if necessary

closed 2023-05-01 by tobozo