tobozo/ESP32-targz #70: Build error Udpate.h missing

olup, 2024-01-09:
Using platformio and:
```
[env:esp32-s3-devkitc-1]
platform = espressif32@6.4.0
board = esp32-s3-devkitc-1
framework = arduino
lib_ldf_mode = deep+
```

I get :
`.pio/libdeps/esp32-s3-devkitc-1/ESP32-targz/src/ESP32-targz-lib.hpp:46:12: fatal error: Update.h: No such file or directory`

tobozo, 2024-01-09:
difficult to say without seeing the code, but here are some tips:

- use `lib_deps = ESP32-targz` in your ini file (otherwise the deep+ ignores it) instead of installing the library manually via vscode
- include <ESP32-targz> from the topmost cpp file, and only include it from there
- try to run 'pio run -e esp32-s3-devkitc-1 -vvv` and check if deep+ is used by the linker, output line should look like this:

`LDF Modes: Finder ~ deep+, Compatibility ~ soft`

otherwise paste the output here, especially the dependency graph

olup, 2024-01-10:
Hey, thanks for your quick response.

I do have `LDF Modes: Finder ~ deep+, Compatibility ~ soft`

I import the lib like this:

```
#define DEST_FS_USES_SD_MMC
#include <ESP32-targz.h>
```

on top of the header file that host this part of the code. Most of my code is in separate compile unit with cpp+h files, but this part is still a simple header included in the main.cpp file - could it be a problem ?

Draft is in this branch https://github.com/olup/grigri/blob/eb98afaba244c506d7d3447f22e7002d91ec355b/software/src/WebServer/WebServer.h#L4-L5

tobozo, 2024-01-10:
I did some actual testing and produced a new release ([1.2.2](https://github.com/tobozo/ESP32-targz/releases/tag/v1.2.2)), it's already available in the platformio library registry.

I have also extended the [platformio example](https://github.com/tobozo/ESP32-targz/blob/master/examples/Test_platformio/platformio.ini) with `lib_ldf_mod=deep+` and a very basic [workflow test](https://github.com/tobozo/ESP32-targz/actions/runs/7480201896/job/20359106554).

```ini
[env:esp32s3]
platform = espressif32
board = esp32-s3-devkitc-1
; Reminder: if using lib_ldf_mode deep+ or chain+, either include the undetected libraries
; from your cpp file, or add them to the lib_deps
lib_ldf_mode = chain+
lib_deps =
  SD
  FS
  SPI
  Update
  LittleFS
  ESP32-targz
```

tobozo, 2024-01-10:
the grigri project fails to compile on linux though, I had to modify file.cpp

https://github.com/olup/grigri/blob/taggz/software/src/file/file.cpp#L1

`#include "./File.h"` should be `#include "./file.h"` (lowercase F)

olup, 2024-01-10:
Great, now version compiles without me having to ad any lib_deps ✅ Thanks for the very quick update

olup, 2024-01-10:
Thanks for the heads up, vscode cpp support is now perfect on refactor/renaming

closed 2024-01-10 by olup