tobozo/ESP32-targz #77: update with encrypted file

AlirezaAbdiyan, 2024-11-01:
Hi,
When eFuse is enabled, the update from memory does not proceed, and there are no error messages either. I used the espsecure.py tool to encrypt my program with a specific key, then stored it along with the SPIFFS as a tar.gz file on the memory. However, the update does not complete successfully. Could you please guide me on what steps I should take to resolve this?

Regards.

tobozo, 2024-11-01:
hi,

thanks for your feedback :+1: 

you may want to try [`GzUpdateClass`](https://github.com/tobozo/ESP32-targz/blob/master/src/ESP32-targz-lib.hpp#L467-L539), this class overloads espressif's [`UpdateClass`](https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/src/Update.h) which should give you more controls on the Update process.

to my knowledge that scenario has only been been tested with esp32FOTA library

here's some pseudo code on how to use it:

```cpp

// get the Update singleton
auto update = GzUpdateClass::getInstance();

// use inherited UpdateClass methods
update.setCryptMode(....); 
update.setCryptKey(....);
update.onProgress( myProgressFunction );

// begin update process
if( !update.begingz(UPDATE_SIZE_UNKNOWN, U_FLASH) )
{
    // GzUpdateClass or UpdateClass error occured
    update.abortz(); // from GzUpdateClass
    return;
}

size_t bytes_written = update.writeGzStream(stream); // from GzUpdateClass

// todo: compare bytes_written with binary size, results may differ (can be aligned to 4096)

if( !update.endz() ) // from GzUpdateClass
{
    // gzip error occured
    return;
}

if( !update.isFinished() ) // from UpdateClass
{
    // UpdateClass error occured
    return;
}

ESP.restart();

```

AlirezaAbdiyan, 2024-11-02:
Thank you for your help. My program code is encrypted but SPIFFS is not! Therefore, I want to extract the tar.gz file from its compressed state onto the SD card first, and then update the files individually. You wrote a function for this, but it seems it cannot extract directly onto the SD card. Could you please guide me on how to use it?

Regards.