tobozo/ESP32-targz #78: feature request : be able to gzip with this library

infrafast, 2024-12-31:
All in the title :-)
Thank you!

Rationale:

The reason I am asking is despite all research I've done, I can't find a way and working library able to gzip data in  file that I can then unzip with ESP32-targz

Since my understanding is that ESP32-targz uses zlib (or uzlib?) , I guess this could be possible, would be very convenient to have all ine one.

tobozo, 2024-12-31:
hi, thanks for your feedback :+1: 

In theory ESP32-targz can decompress content from a web server when `Accept-Encoding: gzip` header was sent with the HTTP request.

Compression support in ESP32-targz has been subject to research for as long as this library exists :wink:
Two possible directions have been identified:

- Activate uzlib builtin compression, requires to add streaming support to [`uzlib_compress(data, src, len)`](https://github.com/tobozo/ESP32-targz/blob/master/src/uzlib/genlz77.c#L98)

- Integrate [zlib's `gzwrite()`](https://github.com/madler/zlib), may require to fully migrate all uzlib code as zlib and uzlib don't coexist well

Any of this requires a lot of focus time, which I don't have at the moment, but this is clearly a feature that will eventually be implemented.

infrafast, 2025-01-01:
Hi thank you very much for having at least considered the feedback ! Understood, this seems to be not straightforward, which also explains why I am struggling to find an existing solution. I would have proposed myself to do it but this seems to be out of my skills. I'll monitor the library evolution.

tobozo, 2025-01-01:
[this branch](https://github.com/tobozo/ESP32-targz/tree/LZ77) can compress using LZ77 (see the [example](https://github.com/tobozo/ESP32-targz/blob/LZ77/examples/ESP32/Deflate_lz77/Deflate_lz77.ino#L70)).

But it's currently limited: while it can write the compressed output to a stream, the uncompressed source must be in memory.

tobozo, 2025-01-02:
other ways to create gzip contents that ESP32-targz can read:

**7zip**
[image]

**gzip**

    gzip -k myfile.txt 

will create a file named:

    myfile.txt.gz

https://gnuwin32.sourceforge.net/packages/gzip.htm
https://git.savannah.gnu.org/cgit/gzip.git

infrafast, 2025-01-02:
Wow, that's what I call an efficient and quick fix :-) !
Works pretty well, with maybe one thing (but it could be because of my setStreamWriter function , please see below

Here is the early test I've done:

compressing the JsonString {"hello":"world","titi":"toto"} into file data.gz

[data.gz](https://github.com/user-attachments/files/18293055/data.gz)

File is ok and readable with 7zip, all good. 

when unzipping using stream, getting : 

 {"hello":"world","titi":"toto"}?␅

Logfile:

dataToCompress->{"hello":"world","titi":"toto"}progress: 0
progress: 9
progress: 12
progress: 16
progress: 19
progress: 22
progress: 25
progress: 29
progress: 32
progress: 35
progress: 38
progress: 41
progress: 45
progress: 48
progress: 51
progress: 54
progress: 58
progress: 61
progress: 64
progress: 67
progress: 70
progress: 74
progress: 77
progress: 80
progress: 90
progress: 93
progress: 96
progress: 100

Compressed 31 input bytes to 31 raw bytes
[GZStreamExpander] Unzipping
[  5920][D][ESP32-targz-lib.cpp:1217] gzUncompress(): gzUncompress( isupdate = false, stream_to_tar = false, use_dict = true, show_progress = false)
[  5925][I][ESP32-targz-lib.cpp:1250] gzUncompress(): [INFO] gzUncompress tradeoff: faster, used 36864 bytes of ram (heap after alloc: 249912)
[  5943][D][ESP32-targz-lib.cpp:1380] gzUncompress(): decompressed 31 bytes

decompressed data ->  {"hello":"world","titi":"toto"}?␅  <-

My setStreamWriter is this one (it is based on dynamic memory allocation that grows as the compression data comes in)

ByteBuffer* uncompressedByteBuffer;

bool streamWriterCallback(unsigned char* buff, size_t buffsize) {
  uncompressedByteBuffer->write(buff, buffsize);
  return true; 
}

struct ByteBuffer {
  uint8_t* buffer = nullptr; 
  size_t size = 0;           
  size_t capacity = 0;     

  // write data in buffer
  void write(const uint8_t* data, size_t len) {
    if (size + len > capacity) {
      // grow
      capacity = (size + len) * 2;
      buffer = (uint8_t*)realloc(buffer, capacity);
    }
    memcpy(buffer + size, data, len); // Copier les données
    size += len;
  }
  ~ByteBuffer() {
    if (buffer) {
      free(buffer);
    }
  }
};

tobozo, 2025-01-03:
I'm quite sure that garbage at the end of the decompressed output is the result of using realloc() while consuming the bytes.

That ByteBuffer class isn't necessary with the latest changes anyway.

```cpp
      uint8_t* dstBuf = NULL;
      size_t dstBufLen = LZPacker::compress( srcBuf, srcBufLen, &dstBuf );
      // do something with dstBuf
      free(dstBuf);
```

infrafast, 2025-01-05:
all right, but I think I will still need my ByteBuffer to decompress, right ?

BTW I performed more test and confirms LZPacker::compress works well, I'd suggest you move it to the main branch (please just give ability to change the progress callback)

tobozo, 2025-01-05:
> I will still need my ByteBuffer to decompress, right?

only if you don't have a filesystem to decompress to

> I'd suggest you move it to the main branch 

it's not ripe yet :)

Release roadmap:
- implement all 4 compression methods
```cpp
    compress( uint8_t* srcBuf, size_t srcBufLen, Stream* dstStream ); // buffer to stream
    compress( uint8_t* srcBuf, size_t srcBufLen, uint8_t** dstBuf );  // buffer to buffer
    compress( Stream* srcStream, size_t srcLen, Stream* dstStream );  // stream to stream
    compress( Stream* srcStream, size_t srcLen, uint8_t* dstBuf );    // stream to buffer
```
- inherit ESP32-targz progress setters, debug level, error handling and filesystem abstraction
- create an example for each compression method
- update readme/documentation
- add new methods to the unit tests (also fix QEmu workflow failing)
- successfully run unit test on all architectures (esp32/esp8266/rp2040)
 
moreover, with compression now available, the next thing is obviously to implement support for creating .tar.gz files :wink: 

this means I'll have to create a tar packer capable of acting as a the source stream in LZPacker::compress() :sweat_smile: 

I can eventually produce a "LZPacker only" release before that, but there's a risk of breaking changes when tar packaging capability will be added.

infrafast, 2025-01-06:
Ok; I can test it when released.

tobozo, 2025-01-06:
Last changes were [tested successfully](https://github.com/tobozo/ESP32-targz/actions/runs/12637935345), so the code is ready for a release :partying_face: 

I'm deliberately ignoring the remaining two points in the roadmap, they won't make sense until I start playing with tar anyway.

> Ok; I can test it when released.

if it's okay I'd love have a last feedback from you about the progress callback setter you were asking for earlier: 

```cpp
  void fn(size_t bytes_read, size_t total_bytes);
  LZPacker::setProgressCallBack( fn );
```

infrafast, 2025-01-08:
Super, thank you I will test and report before week end!

tobozo, 2025-01-08:
I have moved the example sketch to `examples/Test_deflate` folder since it isn't limited to ESP32 any more.

There's also a more complex example that mimics apache's mod_gzip+mod_cache in `examples/ESP32/WebServer_mod_gzip` folder.

infrafast, 2025-01-11:
Hi I've tested the compression of a buffer to a file, please see below my code. My overall programs works and no regression found :-)
The only thing is that for some reason the callback seems not to be called... you can see the log as well.

void gzCallback(size_t bytes_read, size_t total_bytes){
  DEBUG_PRINT("\n ungzip %d bytes ",bytes_read);
}

size_t writeBufferToGzFile(String filename, const char* data){
  DEBUG_PRINT("\nwriteBufferToGzFile->%s",data);  
  File fout = LittleFS.open(filename, "w");
  if(!fout)
  {
    DEBUG_PRINT("\nUnable to create output file, halting");
    return 0;
  }
  LZPacker::setProgressCallBack( gzCallback );
  size_t len = strlen(data);
  size_t outputSize = LZPacker::compress( (uint8_t*)data, len, &fout );
  //size_t fileSize = fout.position();
  DEBUG_PRINT("\nCompressed %d input bytes to %u raw bytes in file%s\n", len, outputSize, filename.c_str());
  fout.close();
  return outputSize;
}

<<<<<<<<<<<<<<<<<<<<<<< LOG 

writeBufferToGzFile->{"interfaces":[{"model":"XR12","protocol":"xair","chSize":12,"busSize":4,"auxSize":1},{"model":"XR16","protocol":"xair","chSize":16,"busSize":4,"auxSize":1},{"model":"XR18","protocol":"xair","chSize":18,"busSize":8,"auxSize":1},{"model":"X32","protocol":"x32","chSize":32,"busSize":8,"auxSize":8},{"model":"MixingStation","chSize":32,"busSize":16,"auxSize":3}],"protocols":[{"name":"xair","ch":{"name":"/ch/config/%d/name","busValue":"","lrValue":""},"bus":{"name":"/bus/config/%d/name","busValue":"","lrValue":""}},{"name":"x32","ch":{"name":"/ch/config/%d/name","busValue":"","lrValue":""},"bus":{"name":"/bus/config/%d/name","busValue":"","lrValue":""}}],"consoles":[{"consoleId":0,"name":"X32/M32"},{"consoleId":2,"name":"X/M AIR"},{"consoleId":6,"name":"Wing"},{"consoleId":13,"name":"HD96-24"},{"consoleId":10,"name":"Avantis"},{"consoleId":11,"name":"dLive"},{"consoleId":4,"name":"GLD"},{"consoleId":17,"name":"iLive"},{"consoleId":5,"name":"SQ"},{"consoleId":1,"name":"Qu"},{"consoleId":16,"name":"CQ"},{"consoleId":8,"name":"Vi"},{"consoleId":3,"name":"Si"},{"consoleId":12,"name":"Ui"},{"consoleId":9,"name":"DL"},{"consoleId":14,"name":"DM"},{"consoleId":15,"name":"TF"},{"consoleId":18,"name":"StudioLive III"}]}
Compressed 1225 input bytes to 420 raw bytes in file/webapp/interfaces.json.gz

tobozo, 2025-01-11:
whoops I switched back to the original uzlib_compress() to implement progress when source=stream and forgot to reattach the progress callback when source=buffer, [I've pushed a fix](https://github.com/tobozo/ESP32-targz/commit/7d48cf5677e68fcef5c2c6210d4de02831f3e27d), sorry about that :blush: 

btw with you might as well compress JSON on the fly, it'll prevent your filesystem from potentially run out of space and it's much faster (you don't have the SPI bottleneck).

that snippet may look a bit dense but it's a summary of what the WebServer_mod_gzip example does: 

```cpp

WebServer(80);

void setup()
{

  server.on("/json", []() {
      int responseCode = 200;
      const char* myJsonData = "{\"ceci\":\"cela\",\"couci\":\"couça\",\"patati\":\"patata\"}";
      server.sendHeader(String(F("Content-Type")), String(F("application/json")), true);
      server.sendHeader(String(F("Content-Encoding")), String(F("gzip")));
      server.sendHeader(String(F("Connection")), String(F("close")));
      // building HTTP response without "Content-Length" header isn't 100% standard, so we have to do this
      String HTTPResponse = String(F("HTTP/1.1"))+' '+String(responseCode)+' '+server.responseCodeToString(responseCode)+"\r\n";
      size_t headersCount = server.responseHeaders();
      for(size_t i=0;i<headersCount;i++)
        HTTPResponse.concat(server.responseHeaderName(i) + F(": ") + server.responseHeader(i) + F("\r\n"));
      HTTPResponse.concat(F("\r\n"));
      // sent HTTP response
      server.client().write(HTTPResponse.c_str(), HTTPResponse.length());
      // stream gz data
      LZPacker::compress( (uint8_t*)myJsonData, strlen(myJsonData), &server.client() );
  });

    server.begin();

}

```

tobozo, 2025-01-11:
btw there's a function to recursively gzip files on a filesystem in the `WebServer_mod_gzip` example, it is disabled to prevent LittleFS from running out of space, but the snippet can be exported:

https://github.com/tobozo/ESP32-targz/blob/7d48cf5677e68fcef5c2c6210d4de02831f3e27d/examples/ESP32/WebServer_mod_gzip/WebServer_mod_gzip.ino#L25-L158

here's how it's used:

```cpp
  // pre-compress all files up to 3 folders deep from the filesystem root
  gzipDir(LittleFS, "/", 3);
```

closed 2025-01-11 by tobozo

infrafast, 2025-01-12:
Thank you very much 1.2.4 works ! 
and the example you provided are very usefull.

tobozo, 2025-01-24:
hi,

Support for creating `.tar` and `.tar.gz` archives has been added to LZ77 branch, see the new [example](https://github.com/tobozo/ESP32-targz/tree/LZ77/examples/Test_tar_packer) and [ongoing discussion](https://github.com/tobozo/ESP32-targz/discussions/80).

infrafast, 2025-02-14:
thank you that's a very good evolution which increases capabilities!