tobozo/ESP32-targz #86: fsTotalBytesCb / fsFreeBytesCb should be uint64_t to support large SD cards

schmitta38, 2025-05-30:
Hi @tobozo, first thank you for your library !

I just faced an issue when trying to unzip some archive using a 32 GB SD card.
gzReadHeader reports wrong free space, I assume this is because fsTotalBytesCb and fsFreeBytesCb are defined as size_t, while the actual space would require using uint64_t (uint32_t is suitable up to approx. 4 GB).

> 23:15:10.940 -> [ 49320][E][LibUnpacker.cpp:1130] gzReadHeader(): [GZ ERROR] Target medium will be out of space (required:1742730743, free:984588288), aborting!
> 23:15:10.940 -> [ 49332][E][LibUnpacker.cpp:1447] gzExpander(): [GZ ERROR] in gzExpander: invalid gzip file or not enough space left on device ?
> 

BTW my archive isn't really 1.7 GB, it's a corrupted download so the Bytes read to get the unpacked data size are nonsense here. But an error could arise in some really cases.

tobozo, 2025-05-31:
hi,

thanks for your feedback 👍 

that check could be disabled using `setupFSCallbacks(nullptr, nullptr)`, but your request makes a lot of sense so I've pushed some changes on branch [1.2.8](https://github.com/tobozo/ESP32-targz/tree/1.2.8)

however this change does not make the library compatible with files>4GB

can you confirm this fixed the issue? if so I'll create a new release

schmitta38, 2025-05-31:
Thanks for your quick fix !

I confirm the available space is now reported correctly 👍 

> 13:39:56.070 -> [ 73876][V][LibUnpacker.cpp:1124] gzReadHeader(): [GZ INFO] valid gzip file detected! gz size: 1075769 bytes, expanded size:3563364784 bytes
> 13:39:56.070 -> [ 73888][V][LibUnpacker.cpp:1133] gzReadHeader(): [GZ INFO] Available space:**31224709120** bytes

The error is the same as when I did not define setupFSCallbacks as you mention, after setting haltOnError(false) :

> 13:40:22.384 -> [100170][V][LibUnpacker.cpp:1321] gzUncompress(): [INFO] Buffer full, now writing 4096 bytes (total=1130496)
> 13:40:22.384 -> [100191][V][LibUnpacker.cpp:1096] gzStreamWriteCallback(): Wrote 4096 bytes
> 13:40:32.443 -> [110231][E][LibUnpacker.cpp:1183] gzReadSourceByte(): gz stream still unresponsive after 10000ms timeout, giving up
> 13:40:32.443 -> uzlib_get_byte/readSourceByte Errors: 1
> 13:40:32.443 -> [110251][E][LibUnpacker.cpp:1480] gzExpander(): gzUncompress returned error code -6
> 13:40:32.490 -> tarGzExpander+intermediate file failed with return code -6

FYI the corrupted download was due to memory allocation, I had some peaks of available data of 16kB on my client, that could not be written at once. After limiting to 4kB blocks, both download + unpack work fine.

[image]

closed 2025-05-31 by tobozo

tobozo, 2025-05-31:
> After limiting to 4kB blocks, both download + unpack work fine.

nice trick with `min(maxBlockSize, stream->available())`, previous implementation was also crashy with malloc when `stream->available() == 0`, I've seen that happen when network timeout value is high and the network is slow

thanks again for your suggestion, you helped make this library better 👍

commit e8d9078a (2025-05-31): support for large SD cards - fixes #86